fhessel/esp32_https_server #184: Meta-issue: is this repo going to be maintained again or should we switch to one of the forks?

jackjansen, 2024-07-07:
It has been a long time since this repo has seen any activity, and I have the feeling that @fhessel is too busy with other things to get back to maintaining it. And the library has completely stopped working with newer versions of IDF and newer boards.

I had a look at the various forks (one by me, @jackjansen , one by @sdrshnptl, one by @jasenk2), but I most liked https://github.com/tmmsunny012/esp32_https_server by @tmmsunny012, which seems to be active at the moment, and which seems to incorporate most of the important pull requests from the other forks.

jackjansen, 2024-07-07:
Hmm, I was a bit too quick, I think. https://github.com/tmmsunny012/esp32_https_server does not seem to be stable at the moment.

tmmsunny012, 2024-07-07:
Hi ,
I forked it 2 days ago and was experimenting on different pull Request. We
can maintain it once we nicely merge the pull request.