fhessel/esp32_https_server #188: SSL Directory not found

teja9121, 2025-04-28:
I'm trying to setup simple HTTPS Server for POST and GET requests. But this is the error I'm facing

ERROR: 
Multiple libraries were found for "WiFi.h"
C:\Users\DELL\Documents\Arduino\libraries\esp32_https_server-master\src/HTTPSServer.hpp:11:10: fatal error: openssl/ssl.h: No such file or directory
 Used: C:\Users\DELL\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0\libraries\WiFi
   11 | #include "openssl/ssl.h"
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board ESP32S3 Dev Module.

**ESP32 S3 Module*
**Software **
 - IDE and Version:  Arduino 1.8.19

Please help me resolving this error !!

dauntless23, 2025-05-14:
i have the same issue, when i try it on arduino ide when i try compiling the sefl signed certificate example, and the  following is my printout

In file included from C:\Users\hkcho\AppData\Local\Temp\.arduinoIDE-unsaved2025414-13852-c9qp5h.s0grp\Self-Signed-Certificate\Self-Signed-Certificate.ino:26:
c:\Users\hkcho\Documents\Arduino\libraries\ESP32_HTTPS_Server\src/HTTPSServer.hpp:11:10: fatal error: openssl/ssl.h: No such file or directory
   11 | #include "openssl/ssl.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1

teja9121, 2025-05-14:
Downgrade the esp32 version to 2.0.6, It's working for me now, please try from yourside

[image]

r57zone, 2025-05-23:
**teja9121,**  This helped! Thank you very much.